mvp/uhubctl #609: Permission denied on Ubuntu 24.04

musahcoding, 2025-02-22:
Hi! 
And tremendous thanks to your wonderful lib 🥇 

On the other hand, I have a semi problem on linux, Ubuntu 24.04. I think the permissions are not taking effect correctly. I first tried to use the `dialout` group, I had the same issue, so I used the open `666` permission. After setting the script here `/etc/udev/rules.d/52-usb.rules` I restarted my PC too.

```
musa@musa-XPS-13-9380:~$ uhubctl -l 1-1.4 -a cycle -p 4
Current status for hub 2-1.4 [17ef:a357 VIA Labs, Inc. USB3.1 Hub, USB 3.10, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Failed to set port status by writing to /sys/bus/usb/devices/2-1.4:1.0/2-1.4-port4/disable (Permission denied).
Follow https://git.io/JIB2Z to make sure that udev is set up correctly.
Falling back to libusb based port control.
Use -S to skip trying the sysfs interface and printing this message.
Sent power off request
New status for hub 2-1.4 [17ef:a357 VIA Labs, Inc. USB3.1 Hub, USB 3.10, 4 ports, ppps]
  Port 4: 00a0 off
Current status for hub 1-1.4 [17ef:1029 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0503 power highspeed enable connect [12d1:1506 HUAWEI HUAWEI Mobile]
Failed to set port status by writing to /sys/bus/usb/devices/1-1.4:1.0/1-1.4-port4/disable (Permission denied).
Follow https://git.io/JIB2Z to make sure that udev is set up correctly.
Falling back to libusb based port control.
Use -S to skip trying the sysfs interface and printing this message.
Sent power off request
New status for hub 1-1.4 [17ef:1029 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0000 off
Current status for hub 2-1.4 [17ef:a357 VIA Labs, Inc. USB3.1 Hub, USB 3.10, 4 ports, ppps]
  Port 4: 00a0 off
Failed to set port status by writing to /sys/bus/usb/devices/2-1.4:1.0/2-1.4-port4/disable (Permission denied).
Follow https://git.io/JIB2Z to make sure that udev is set up correctly.
Falling back to libusb based port control.
Use -S to skip trying the sysfs interface and printing this message.
Sent power on request
New status for hub 2-1.4 [17ef:a357 VIA Labs, Inc. USB3.1 Hub, USB 3.10, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.4 [17ef:1029 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0000 off
Failed to set port status by writing to /sys/bus/usb/devices/1-1.4:1.0/1-1.4-port4/disable (Permission denied).
Follow https://git.io/JIB2Z to make sure that udev is set up correctly.
Falling back to libusb based port control.
Use -S to skip trying the sysfs interface and printing this message.
Sent power on request
New status for hub 1-1.4 [17ef:1029 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0100 power
```
Also the dialout group (`musa`) is my user:
```
musa@musa-XPS-13-9380:~$ getent group dialout
dialout:x:20:musa
```

And the current content of `/etc/udev/rules.d/52-usb.rules`:
```
musa@musa-XPS-13-9380:~$ cat /etc/udev/rules.d/52-usb.rules
SUBSYSTEM=="usb", DRIVER=="usb", MODE="0666"
# Linux 6.0 or later (its ok to have this block present for older Linux kernels):
SUBSYSTEM=="usb", DRIVER=="usb", \
  RUN="/bin/sh -c \"chmod -f 666 $sys$devpath/*-port*/disable || true\""
```

**The problem:** As you can see in the output, I get the permission denied using my own user, and the command to turn off the usb works at 2 try and the turn on also at 2nd try. 

The `cycle` works on the first try even I get permission denied.

Also with `sudo` everything works as expected.

**Desired behaviour:** I would like to get the tool working without `sudo` reliably.

Thanks!

musahcoding, 2025-02-22:
The kernel version:
```
musa@musa-XPS-13-9380:~$ uname -r
6.8.0-53-generic
```

musahcoding, 2025-02-22:
Also I noticed:

The device I am operating on is a usb modem. When I use `sudo` the device is disconnected (`lsusb` does not see it) and **the small green light of the modem is off too** but without `sudo` and the permission denied problem, the device is gone from `lsusb` but the light of the modem keep blinking

mvp, 2025-02-23:
Fixed in 42cdfb0 - please update udev rule to use `*port*`.

closed 2025-02-23 by mvp