mvp/uhubctl #614: uhubctl not working in Raspberry PI 4B + HAOS

polmonta, 2025-04-29:
Hello,
thank you for this, it looks very promising, but I am unable to make it work.

I'm trying to shut down the USB ports in the RPI, but I'm constantly getting the "No compatible devices detected at location 2!" error.

To use uhubctl in HAOS, I've used [this](https://github.com/mochipon/hassio-uhubctl) addon, which I managed to set up properly and works. I'm now using the addon from the Terminal (using the SSH addon) to further debug. The USB device is connected to a 3.0 port, but I don't mind shuttind down all USB ports.

I've upgraded the VL805 firmware to 138C0 by loading another SD card to the RPI with the bootloader. 

Here's what I'm getting:

```
local-uhubctl:/app# uhubctl 
Current status for hub 1 [1d6b:0002 Linux 6.6.74-haos-raspi xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ganged]
Current status for hub 2 [1d6b:0003 Linux 6.6.74-haos-raspi xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [152d:0578 JMicron USB to ATA/ATAPI Bridge 0123456789ABCDEF]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
```

```
local-uhubctl:/app# uhubctl -a 0 -l 2
No compatible devices detected at location 2!
Run with -h to get usage info.
```

```
local-uhubctl:/app# uhubctl -l 1-1 -a 0
No compatible devices detected at location 1-1!
Run with -h to get usage info.
```

```
local-uhubctl:/app# lsusb -t
Bus 002 Device 003: ID 152d:0578 JMicron USB to ATA/ATAPI Bridge
Bus 001 Device 001: ID 1d6b:0002 Linux 6.6.74-haos-raspi xhci-hcd xHCI Host Controller
Bus 001 Device 008: ID 2109:3431 USB2.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux 6.6.74-haos-raspi xhci-hcd xHCI Host Controller
```

I've even added these:
```
"privileged": ["SYS_ADMIN", "SYS_RAWIO"],
"devices": ["/dev/bus/usb"]
```
to the config.json of the addon but that didn't help.

Any ideas would be greatly appreciated.

polmonta, 2025-04-29:
I'm also adding my uhubctl version:
```
local-uhubctl:/app# uhubctl -v
2.6.0-24-gfbe812bd
```

mvp, 2025-05-01:
Can you please show output for running
```
sudo uhubctl -f
```
Also, what is your os/kernel version? Just show `uname -a` and `lsb_release -a`.

mvp, 2025-05-10:
Can you please show output of

```
cat /sys/firmware/devicetree/base/model
```
If this is different from `Raspberry Pi 4 Model B`, I should be able to provide a fix.

Basically, current code logic is to check if hub `1-1` has vendor:model as `2109:3431` - which your hub `1-1` attached to hub `1` does:

```
Current status for hub 1 [1d6b:0002 Linux 6.6.74-haos-raspi xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ganged]
```

But it also checks if current computer model as read from ` /sys/firmware/devicetree/base/model` is `Raspberry Pi 4 Model B`. If it reads differently, it will not work. But current code can be patched to recognize different variations if needed.

polmonta, 2025-05-12:
I tried to run
`cat /sys/firmware/devicetree/base/model`
but HomeAssistant can't find the file, neither in the HA terminal nor inside the uhubctl add-on's terminal (container).
However, I was able to see the HW version of the Raspberry Pi through the UI.

[image]

It also shows info on the usbs, if that helps:
usb1:
`BUSNUM: '001'
DEVNAME: /dev/bus/usb/001/001
DEVNUM: '001'
DEVPATH: /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1
DEVTYPE: usb_device
DRIVER: usb
ID_BUS: usb
ID_FOR_SEAT: usb-platform-fd500000_pcie-pci-0000_01_00_0
ID_MODEL: xHCI_Host_Controller
ID_MODEL_ENC: xHCI\x20Host\x20Controller
ID_MODEL_ID: '0002'
ID_PATH: platform-fd500000.pcie-pci-0000:01:00.0
ID_PATH_TAG: platform-fd500000_pcie-pci-0000_01_00_0
ID_REVISION: '0606'
ID_SERIAL: Linux_6.6.74-haos-raspi_xhci-hcd_xHCI_Host_Controller_0000:01:00.0
ID_SERIAL_SHORT: '0000:01:00.0'
ID_USB_INTERFACES: ':090000:'
ID_USB_MODEL: xHCI_Host_Controller
ID_USB_MODEL_ENC: xHCI\x20Host\x20Controller
ID_USB_MODEL_ID: '0002'
ID_USB_REVISION: '0606'
ID_USB_SERIAL: Linux_6.6.74-haos-raspi_xhci-hcd_xHCI_Host_Controller_0000:01:00.0
ID_USB_SERIAL_SHORT: '0000:01:00.0'
ID_USB_VENDOR: Linux_6.6.74-haos-raspi_xhci-hcd
ID_USB_VENDOR_ENC: Linux\x206.6.74-haos-raspi\x20xhci-hcd
ID_USB_VENDOR_ID: 1d6b
ID_VENDOR: Linux_6.6.74-haos-raspi_xhci-hcd
ID_VENDOR_ENC: Linux\x206.6.74-haos-raspi\x20xhci-hcd
ID_VENDOR_ID: 1d6b
MAJOR: '189'
MINOR: '0'
OF_COMPATIBLE_N: '0'
OF_FULLNAME: /scb/pcie@7d500000/pci@0,0/usb@0,0
OF_NAME: usb
PRODUCT: 1d6b/2/606
SUBSYSTEM: usb
TAGS: ':seat:'
TYPE: 9/0/1
USEC_INITIALIZED: '3392444'`

usb2:
`BUSNUM: '002'
DEVNAME: /dev/bus/usb/002/001
DEVNUM: '001'
DEVPATH: /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2
DEVTYPE: usb_device
DRIVER: usb
ID_BUS: usb
ID_FOR_SEAT: usb-platform-fd500000_pcie-pci-0000_01_00_0
ID_MODEL: xHCI_Host_Controller
ID_MODEL_ENC: xHCI\x20Host\x20Controller
ID_MODEL_ID: '0003'
ID_PATH: platform-fd500000.pcie-pci-0000:01:00.0
ID_PATH_TAG: platform-fd500000_pcie-pci-0000_01_00_0
ID_REVISION: '0606'
ID_SERIAL: Linux_6.6.74-haos-raspi_xhci-hcd_xHCI_Host_Controller_0000:01:00.0
ID_SERIAL_SHORT: '0000:01:00.0'
ID_USB_INTERFACES: ':090000:'
ID_USB_MODEL: xHCI_Host_Controller
ID_USB_MODEL_ENC: xHCI\x20Host\x20Controller
ID_USB_MODEL_ID: '0003'
ID_USB_REVISION: '0606'
ID_USB_SERIAL: Linux_6.6.74-haos-raspi_xhci-hcd_xHCI_Host_Controller_0000:01:00.0
ID_USB_SERIAL_SHORT: '0000:01:00.0'
ID_USB_VENDOR: Linux_6.6.74-haos-raspi_xhci-hcd
ID_USB_VENDOR_ENC: Linux\x206.6.74-haos-raspi\x20xhci-hcd
ID_USB_VENDOR_ID: 1d6b
ID_VENDOR: Linux_6.6.74-haos-raspi_xhci-hcd
ID_VENDOR_ENC: Linux\x206.6.74-haos-raspi\x20xhci-hcd
ID_VENDOR_ID: 1d6b
MAJOR: '189'
MINOR: '128'
OF_COMPATIBLE_N: '0'
OF_FULLNAME: /scb/pcie@7d500000/pci@0,0/usb@0,0
OF_NAME: usb
PRODUCT: 1d6b/3/606
SUBSYSTEM: usb
TAGS: ':seat:'
TYPE: 9/0/3
USEC_INITIALIZED: '3317145'`

mvp, 2025-05-13:
It seems that your OS does not support devicetree yet. Modern Linux distros are all using devicetree, which creates subtree `/sys/firmware/devicetree`, and most importantly `/sys/firmware/devicetree/base/model`.

From what I know about non-devicetree distros, they all provide some version of model in `/proc/cpuinfo` instead. Can you please show output of `cat /proc/cpuinfo`?

And of course, using devicetree based distro would fix it too.

polmonta, 2025-05-13:
You're right. `cat /proc/cpuinfo` did it:
```
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Revision	: b03114
Serial		: 10000000c421a6d5
Model		: Raspberry Pi 4 Model B Rev 1.4
```

I will let the HomeAssistant devs know about devicetree. Thanks!

mvp, 2025-05-14:
Thanks for confirming this. I will provide a fix after I return from a trip about 2 weeks from now. In meantime, you can use `uhubctl -f` as a workaround: this forces operation on non-compliant hubs.

polmonta, 2025-05-17:
I have some updates. AI got it to work by modifying part of the code to add some RPI flags. However, the drive turns back on after a couple of minutes. I've tried using the -r N flag, but no matter what N I set, the device comes back online. 
I tried the other options but I didn't have any success either.

Here's the command I need to execute to get it working:
```
git clone --depth 1 https://github.com/mvp/uhubctl.git /tmp/uhubctl \
 && sed -i 's/static int is_rpi_4b = 0;/static int is_rpi_4b = 1;/' /tmp/uhubctl/uhubctl.c \
 && sed -i 's/if (rc == 0 && desc.bDeviceClass != LIBUSB_CLASS_HUB)/if (0)/' /tmp/uhubctl/uhubctl.c \
 && sed -i 's/if (info.lpsm != HUB_CHAR_INDV_PORT_LPSM && !opt_force)/if (0)/' /tmp/uhubctl/uhubctl.c \
 && sed -i 's/is_rbi_4b = check_computer_model("Raspberry Pi 4 Model B");//' /tmp/uhubctl/uhubctl.c \
```

commit b3fa9e7e (2025-05-31): Improve computer model detection when device tree is not available

Closes #614.
closed 2025-05-31 by mvp