mvp/uhubctl #615: Let --search override --nodesc

emsaks, 2025-05-05:
I use `uhubctl` within a function called by other parts of a script, that can pass parameters to be forwarded to `uhubctl`. I like to use `--nodesc` by default, so the 'inner' function specifies it, but sometimes the function caller needs to search by description. Obviously, if the `--search` parameter is specified, the is no choice but to query device description, so the `--nodesc` should be ignored.

mvp, 2025-05-07:
Not trying to dismiss a need for proposed change, but:
 
* It is still technically possible for `--search` and `--searchhub` to work even without querying descriptor strings. E.g. one might want to limit to only 2 port USB hub using `uhubctl --nodesc --searchhub "2 ports"`, and it would work correctly. Or, e.g. one can call `uhubctl --nodesc --search "0bda:0411"`, even though it is equivalent to `uhubctl --nodesc --vendor 0bda:0411`.
* What stops you from changing your script invocation logic to be smarter and remove `--nodesc` from parameters if `--search` is specified?

emsaks, 2025-05-08:
That's a good point. I didn't realize `--search` included more than the device descriptor. I *am* using the script logic you mentioned, as a workaround. It just complicates what would have been a simple function. Thanks.

mvp, 2025-05-10:
Thanks, I will close this then!

closed 2025-05-10 by mvp