mvp/uhubctl #616: New working USB hub entry

joelpmichael, 2025-05-17:
RSHTECH 10-in-1, available on AliExpress: https://www.aliexpress.com/item/1005008160865046.html

All ports work with uhubctl on MacOS 15 except second USB-C 3.2 port. Internally configured as 4-port USB 3.2 hub, daisy-chained to 4-port USB-3.0 hub, daisy-chained to 4-port USB-3.0 hub.

mvp, 2025-05-17:
Can you please provide output for `uhubctl`? Also, how did you test vbus support? Very few Rosonway hubs actually support vbus control. All too often, they just turn off data lines, but not voltage.

joelpmichael, 2025-05-17:
Yep can do, is there anything in particular you’d like me to test? The two USB-A ports I’m using definitely cut VBUS because I use them to power cycle the board I’m developing on and/or the hardware debugger that crashes. -- jpm

commit a05a2506 (2025-05-31): Add Rosonway RSH-ST10C-6 as compatible device

Closes #616.
closed 2025-05-31 by mvp