mvp/uhubctl #617: Uhubctl - different behavior between Android phones and iPhones

ICE-FlatLine, 2025-05-22:
Hi,
So I’ve been playing around with uhubctl and Amazon basics 10-port USB hub, and I have noticed some inconsistency.

My host computer - Macbook Pro M3 (Apple Silicon)

When toggling power for USB ports with Android phones connected (I am using primarily Samsung Galaxy S22-S24 devices), it works as expected: port power-off -> device disconnects from host and stops charging.

However, when doing the same with iPhones connected, only data connection is cut off and iPhone is still charging. Physically disconnecting usb cable and reconnecting back -> iPhone no longer charging.

Anybody encountered something similar? 
Am I missing something obvious here?

Thanks in advance.

mvp, 2025-05-24:
It is possible that your Amazon Basics hub doesn't truly support turning vbus off, and uhubctl is only able to turn off data lines (but uhubctl itself cannot tell this, since hub chip reports as supporting per port power switching). It looks like android is smart enough to voluntarily stop charging once it detects data lines off. But iPhone seems to be treating hub as dumb charger and is still drawing power regardless.

I recommend trying another hub from supporter list.

mvp, 2025-05-31:
Sorry, I will close this as this is a hardware issue. Not much can be done by software here (uhubctl).
I recommend looking for another hub...

closed 2025-05-31 by mvp

ICE-FlatLine, 2025-05-31:
Thank you,
I will indeed try a different hub