mvp/uhubctl #618: Linux permissions error with current udev rules, fixed by specifying DRIVER=="hub"

grvvy, 2025-05-28:
I mentioned finding a mismatch between the readme and the `udev/rules.d/52-usb.rules` DRIVER type in #574 and the fix was to change it from `DRIVER=hub` to `DRIVER=usb`.

When I try running a uhubctl port toggling command  with this change, I get a permissions error:

```
$ $ uhubctl --location 1-2.3 --port 3 -a off
Current status for hub 2-2.3 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 3: 02a0 power 5gbps Rx.Detect
Failed to set port status by writing to /sys/bus/usb/devices/2-2.3:1.0/2-2.3-port3/disable (Permission denied).
Follow https://git.io/JIB2Z to make sure that udev is set up correctly.
Falling back to libusb based port control.
Use -S to skip trying the sysfs interface and printing this message.
Sent power off request
New status for hub 2-2.3 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 3: 00a0 off
Current status for hub 1-2.3 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0503 power highspeed enable connect [1fc9:000c NXP LPC ABCD]
Sent power off request
New status for hub 1-2.3 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0000 off
```
but just running a plain `uhubctl` command to list all hubs/devices works just fine:
```
$ uhubctl
Current status for hub 2-2.3 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 00a0 off
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-2 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 0263 power 5gbps U3 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-3 [05e3:0610 USB2.0 Hub, USB 2.00, 2 ports, ppps]
  Port 1: 0103 power enable connect [10ab:9309]
  Port 2: 0100 power
Current status for hub 1-2.3 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect [1d50:6089 Great Scott Gadgets HackRF One 0000000000000000325866e6293d4423]
  Port 3: 0000 off
  Port 4: 0100 power
Current status for hub 1-2 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0100 power
```
and when I revert it back to `DRIVER=hub`, I no longer get the error message when trying to toggle a port.

I don't know a ton about udev or device management in Linux, could I get some insight as to what's happening here?

I'm using `Ubuntu 24.04.2 LTS` with kernel version `Linux 6.11.0-26-generic` and `uhubctl-2.6.0-24-gfbe812bd` and my current user is in the dialout group.

udev rule:
```
# This is for Linux before 6.0:
SUBSYSTEM=="usb", DRIVER=="usb", MODE="0664", GROUP="dialout"

# This is for Linux 6.0 or later (ok to keep this block present for older Linux kernels):
SUBSYSTEM=="usb", DRIVER=="usb", \
  RUN+="/bin/sh -c \"chown -f root:dialout $sys$devpath/*port*/disable || true\"" \
  RUN+="/bin/sh -c \"chmod -f 660 $sys$devpath/*port*/disable || true\""
  RUN+="/usr/local/bin/udev_trigger_test.sh"
```

mvp, 2025-05-31:
I am not really sure what's going on here. It seems like some linux distros have DRIVER as `hub`, and some as `usb`. I don't see a specific pattern, e.g. on Raspberry Pi 5 running most recent bookworm and kernel 6.12.15 I observe driver as `usb`.

Perhaps we need to duplicate those rules? Unfortunately, udev does allow multimatch on single udev rule to make it simple.

grvvy, 2025-06-05:
Sorry for the delay, that does seem to work on my system. Thank you! Don't forget the dangling test script run I placed in there at the end. Thanks for all your work!

commit 786dcb22 (2025-06-06): Improve Linux udev rules to support more systems

Some Linux systems detect USB hubs with driver as `hub`, and some as `usb`.
Improve udev permission rules to match both options.

Closes #618.
closed 2025-06-06 by mvp