mvp/uhubctl #620: tested Targus ACH155, j5create JUH340

b-soder, 2025-06-04:
Targus ACH155, power is controllable on port 3, turns on/off all four ports

j5create JUH340, power is controllable for port 1 alone, 2/3/4 are controlled by port 2

bought both at Staples, along with 3-4 others that don't work at all.

mvp, 2025-06-05:
jCreate JUH340 was already rejected in #294 because it was detected as ganged back then.
Targus ACH155 seems to be ganged too?
Can you please show output of uhubctl with these hubs connected? Thanks!

b-soder, 2025-06-05:
I returned the Targus already, sorry. here's the 340. it's separate for port 1, 2/3/4 are connected.

Current status for hub 1-1 [05e3:0610 GenesysLogic USB2.1 Hub, USB 2.10, 4 ports, ppps]

closed 2025-06-06 by mvp

mvp, 2025-06-06:
Added Targus. I remain skeptical about jCreate JUH340 simply because apparently the same model name appears to be quite different from what was reported in #294 5 years ago.

Do you see USB3 counterpart in the list?
You should be able to see something like this:
```
Current status for hub 1-1 [05e3:0610 GenesysLogic USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 2-1 [05e3:0610 GenesysLogic USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
```

b-soder, 2025-06-07:
Yeah, I just checked, the cheap usb to microusb adapter I'm using to connect this to my pi zero is 2.0 only. Connecting it to my desktop (mint), it's showing more there.

Current status for hub 4-3.3 [05e3:0620 GenesysLogic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 3-3.3 [05e3:0610 GenesysLogic USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power

however, when I turn them all off, there's still power on all four ports, no data though, phone connects and disconnects when I toggle the setting. I tried adding -r 5, power is still on:  sudo uhubctl -p4 -a1 -l 3-3.3 -r 5

Anything else I can try? Power toggling works on the pi but not on the desktop.

mvp, 2025-06-07:
Thanks for more testing - that means it does not support vbus off.
I am trying very hard to make sure we only list devices that do support vbus off.

b-soder, 2025-06-07:
why does it work on the pi? any idea?

mvp, 2025-06-07:
Raspberry Pi supports vbus off correctly (but each model has different way to do it).
Are you sure you were turning off ports on JUH340, not on Raspberry Pi root hub itself?

b-soder, 2025-06-08:
from what I've found, the zero2w only has a single 5v rail that is connected to all USB ports and isn't controllable. Different from other models, there's a reason it costs 20 bucks. :)

In any case, I was definitely turning off ports on the hub. I had something connected to 1 and 2, and they are independent. 2/3/4 aren't.

reshef-rt, 2025-08-11:
Tested with the Targus ACH155, with an iphone on USB A -> USB-C charging. `uhubctl -a toggle` does not seem to affect vbus (phone stays charging).

It doesn't seem compatible.

Expected: iPhone stops charging
Experienced: iPhone keeps charging even while ports are marked as "off" on Targus ACH155, USB data properly disconnects when ports are off.

```
reshef@Reshefs-MBP ~ % uhubctl
Current status for hub 1-2 [2109:0817 VIA Labs, Inc. USB3.0 Hub 000000000, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [05ac:12a8 Apple Inc. iPhone 00008130000478502EA0001C]
reshef@Reshefs-MBP ~ % uhubctl -a off
Current status for hub 1-2 [2109:0817 VIA Labs, Inc. USB3.0 Hub 000000000, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 1-2 [2109:0817 VIA Labs, Inc. USB3.0 Hub 000000000, USB 3.20, 4 ports, ppps]
  Port 1: 00a0 off
  Port 2: 00a0 off
  Port 3: 00a0 off
  Port 4: 00a0 off
Current status for hub 1-1 [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [05ac:12a8 Apple Inc. iPhone xxxxxxxxx]
Sent power off request
New status for hub 1-1 [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
reshef@Reshefs-MBP ~ % #iphone still charging
reshef@Reshefs-MBP ~ % uhubctl -a on
Current status for hub 1-2 [2109:0817 VIA Labs, Inc. USB3.0 Hub 000000000, USB 3.20, 4 ports, ppps]
  Port 1: 00a0 off
  Port 2: 00a0 off
  Port 3: 00a0 off
  Port 4: 00a0 off
Sent power on request
New status for hub 1-2 [2109:0817 VIA Labs, Inc. USB3.0 Hub 000000000, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
Sent power on request
New status for hub 1-1 [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0101 power connect [05ac:12a8]
```

commit fab805bc (2025-06-06): Add Targus ACH155 as compatible device

Per submitter, Targus ACH155 port 3 turns on/off power on all four ports.

Closes #620.