mvp/uhubctl #622: All commands hang

hyok2025, 2025-06-11:
sw_vers:
ProductName:		macOS
ProductVersion:		14.4
BuildVersion:		23E214

uhubctl version 2.6.0

Running `uhubctl` with/without arguments hangs forever (except uhubctl --hep).
Any ways to debug?

hyok2025, 2025-06-11:
[image]
it hangs here. Any way to tell which usb port caused this from the desc?
Appreciate if you can provide some help.

hyok2025, 2025-06-11:
More info:
It was working when i tested it manually. But after i integrated uhubctl with my python server which can trigger multiple uhubctl -a cycle calls (with unique port arguments) it started getting this issue.

mvp, 2025-06-11:
It seems that it hangs trying to get hub descriptors. However, this control transfer is set to have 5 sec timeout which apparently is not honored.

Usually this is caused by non-compliant USB device. Or, your hub may have wedged into unresponsive state when you tried to send power commands in parallel. Theoretically, OS and libusb should be able to not freeze even if USB device is not compliant. However, in practice it is hard to achieve, and bad external devices can still cause big trouble.

hyok2025, 2025-06-11:
Is it possible to know which external device (or which port) is causing trouble using other tools considering the uhubctl just hang. Thanks in advance.

mvp, 2025-06-11:
I don't know. Note that hang is happening in libusb, which is likely calling Darwin kernel. You might be able to understand in which syscall it is hanging using dtruss (Mac strace equivalent).

May be good idea to check which libusb version you are linking with and report this in libusb.

closed 2025-06-12 by hyok2025